joshbenner/esphome-daikin-s21 #3: Status bounces in HA

joshbenner, 2023-06-02:
When changing a setting, such as setpoint or mode via HA UI, the state often bounces to its previous version before reflecting the updated state. I suspect this is because either the ESPHome device was already in the process of gathering and sending a state update prior to receiving a command, or the HVAC unit lags in its state reporting by a second or two.

Idea: When receiving a command, update internal state and temporarily delay interval-based state updates.

joshbenner, 2023-06-10:
Fixed in 1.2.1

closed 2023-06-10 by joshbenner

daniel-dona, 2024-08-17:
This is still happening, mostly for actions that take some time to complete by the AC unit (turn on). Check this log, "Mode: COOL" is received from HA, but at least 2 additional status reports as "Mode: FAN_ONLY" are sent back. In HA the UI immediately reports the change to cooling, then it bouces back to fan only and then bounces again to cooling. This is specially problematic if during the bounces you change other property, then states can end mixed...

```
[08:09:37][I][daikin_s21.climate:409]: Controlling S21 climate:
[08:09:37][I][daikin_s21.climate:410]:   Mode: COOL
[08:09:37][I][daikin_s21.climate:412]:   Setpoint: 26.0 (s21: 26.0)
[08:09:37][I][daikin_s21.climate:413]:   Fan: Automatic
[08:09:37][D][daikin_s21:415]: Sending basic climate CMD (D1): 13PA
[08:09:37][D][climate:396]: 'Daikin AC' - Sending state:
[08:09:37][D][climate:399]:   Mode: FAN_ONLY
[08:09:37][D][climate:401]:   Action: IDLE
[08:09:37][D][climate:407]:   Custom Fan Mode: Automatic
[08:09:37][D][climate:416]:   Swing Mode: VERTICAL
[08:09:37][D][climate:419]:   Current Temperature: 27.00°C
[08:09:37][D][climate:425]:   Target Temperature: 26.00°C
[08:09:37][W][component:237]: Component api took a long time for an operation (695 ms).
[08:09:37][W][component:238]: Components should block for at most 30 ms.
[08:09:38][W][component:237]: Component daikin_s21 took a long time for an operation (597 ms).
[08:09:38][W][component:238]: Components should block for at most 30 ms.
[08:09:39][W][component:237]: Component daikin_s21 took a long time for an operation (599 ms).
[08:09:39][W][component:238]: Components should block for at most 30 ms.
[08:09:40][V][sensor:043]: 'Temperatura interior': Received new state 27.000000
[08:09:40][D][sensor:094]: 'Temperatura interior': Sending state 27.00000 °C with 1 decimals of accuracy
[08:09:41][V][sensor:043]: 'Temperatura exterior': Received new state 26.500000
[08:09:41][D][sensor:094]: 'Temperatura exterior': Sending state 26.50000 °C with 1 decimals of accuracy
[08:09:41][V][sensor:043]: 'Temperatura serpentín': Received new state 26.500000
[08:09:41][D][sensor:094]: 'Temperatura serpentín': Sending state 26.50000 °C with 1 decimals of accuracy
[08:09:41][V][sensor:043]: 'Revoluciones ventilador': Received new state 690.000000
[08:09:41][D][sensor:094]: 'Revoluciones ventilador': Sending state 690.00000 rpm with 0 decimals of accuracy
[08:09:41][D][climate:396]: 'Daikin AC' - Sending state:
[08:09:41][D][climate:399]:   Mode: FAN_ONLY
[08:09:41][D][climate:401]:   Action: IDLE
[08:09:41][D][climate:407]:   Custom Fan Mode: Automatic
[08:09:41][D][climate:416]:   Swing Mode: VERTICAL
[08:09:41][D][climate:419]:   Current Temperature: 27.00°C
[08:09:41][D][climate:425]:   Target Temperature: 26.00°C
[08:09:41][W][component:237]: Component daikin_s21 took a long time for an operation (599 ms).
[08:09:41][W][component:238]: Components should block for at most 30 ms.
[08:09:43][W][component:237]: Component daikin_s21 took a long time for an operation (596 ms).
[08:09:43][W][component:238]: Components should block for at most 30 ms.
[08:09:45][W][component:237]: Component daikin_s21 took a long time for an operation (596 ms).
[08:09:45][W][component:238]: Components should block for at most 30 ms.
[08:09:45][D][climate:396]: 'Daikin AC' - Sending state:
[08:09:46][D][climate:399]:   Mode: COOL
[08:09:46][D][climate:401]:   Action: IDLE
[08:09:46][D][climate:407]:   Custom Fan Mode: Automatic
[08:09:46][D][climate:416]:   Swing Mode: OFF
[08:09:46][D][climate:419]:   Current Temperature: 27.00°C
[08:09:46][D][climate:425]:   Target Temperature: 26.00°C
```