joshbenner/esphome-daikin-s21 #4: Temperature can go lower?

joshbenner, 2023-06-06:
Faikin can now go down to 14 C -- can esphome-daikin-s21 do the same?

https://github.com/revk/ESP32-Faikin/issues/41#issuecomment-1578070463

pfak, 2023-07-22:
This would be amazing (and I would have to pay someone to build these boards for me).

I have an issue where my Daikin thinks it's cool enough (because of where the thermistor is located), and the unit refuses to cool more. I have been getting around this by running the fans on the units, but this blows humidity back into the room.

I would love functionality like this!

TheKayneGame, 2024-01-14:
Yes it can and it works.
I tried adding:
```
    visual:
      min_temperature: 10
      max_temperature: 30
```
to my climate component and it looks like it actually sets that temperature