joshbenner/esphome-daikin-s21 #8: Upgrading from ESP-Faikin

XMaarten, 2023-07-08:
Can you add instructions on how to get the ESPHome code on the device to the README please? 

Revk explained this on https://github.com/revk/ESP32-Faikin/issues/69

joshbenner, 2023-07-08:
Programming devices is covered by the [ESPHome ](https://esphome.io/) documentation. This code is not specific to Faikins.

Also, I do not provide pre-compiled binaries that would be usable to flash from Faikin's OTA upgrade mechanism -- the binaries produced for ESPHome installs are intended to be unique/specific to the individual device in most cases and compiled/installed from ESPHome itself, so providing pre-compiled binaries is not generally done with ESPHome.

closed 2023-07-08 by joshbenner

XMaarten, 2023-07-08:
I know how to program ESPHome devices, but the Faikin device does not have a USB interface or easy terminals to use. Being able to load an ESPHome binary on it through WiFi makes it a lot easier. You don't need a pre-compiled binary, you can download the binary from the ESPHome webinterface.