joshbenner/esphome-daikin-s21 #14: Wiring for ESP32

Nik19792000, 2023-10-02:
Hi,
first of all: great work to bring Faikin to ESPHome!
I'm planning to use a standard [ESP32](https://www.aliexpress.com/item/32864722159.html?spm=a2g0o.order_list.order_list_main.5.26051802nNZSjv) board to connect Daikin's S21 port via a [voltage step down](https://www.aliexpress.com/item/1005002168374063.html?spm=a2g0o.productlist.main.7.7b8f23a86sQ0uf&algo_pvid=7c8af159-e753-48f1-a638-d9a8908e95ce&algo_exp_id=7c8af159-e753-48f1-a638-d9a8908e95ce-3&pdp_npi=4%40dis%21SGD%211.48%211.3%21%21%211.05%21%21%402140e03216962523912442075ea1ea%2112000018924816895%21sea%21SG%21164400281%21&curPageLogUid=mqjxvyEnPKlt).
Does the following wiring diagram look correct or do I need any resistors?
Is there a difference, if I step down to 5V or 3.3V?
[image]

joshbenner, 2023-10-02:
I used revk's Faikin board because this type of stuff is not a skillset I'm confident in, so I'm really not sure how best to answer your question.

I can say that when I discussed some of this on the ESPHome discord, they felt that a voltage shift wasn't necessary, but I've never confirmed that myself. You may get much better input from @revk or from the ESPHome discord.

I'm sorry I cannot be of much help here -- maybe someone else can chime in?

revk, 2023-10-02:
My Faikin handles the voltage shift both ways because it was necessary on some models. But yes, on some models it just worked showing 3.3V GPIOs on it each way.

Nik19792000, 2023-10-03:
Thanks for your responses!
@revk so, if I understand you right, then some models may work according to my diagram and some may need further handling of RX/TX with resistors, correct?

revk, 2023-10-03:
Correct

closed 2023-10-03 by Nik19792000

Nik19792000, 2023-11-11:
Quick update: For my Daikin FTKD60JVE it works perfectly with above wiring and the default config mentioned here: https://github.com/joshbenner/esphome-daikin-s21#configuration-example