joshbenner/esphome-daikin-s21 #15: ESP32 board can't receive the state of the AC

meditation1985, 2023-11-24:
I'm  using a standard [ESP32] board to connect Daikin's S21 (ftka20bv1h) with default config mentioned here: https://github.com/joshbenner/esphome-daikin-s21#configuration-example

but keep show the following error.
Sending poweron commond is work but can't receive the state of the AC.

21:31:47	[W]	[component:214]	
Component daikin_s21 took a long time for an operation (0.90 s).
21:31:47	[W]	[component:215]	
Components should block for at most 20-30ms.
21:31:48	[E]	[uart:015]	
Reading from UART timed out at byte 0!
21:31:48	[W]	[daikin_s21:257]	
Timeout waiting for F1 response
21:31:49	[E]	[uart:015]	
Reading from UART timed out at byte 0!
21:31:49	[W]	[daikin_s21:257]	
Timeout waiting for F5 response
21:31:49	[E]	[uart:015]	
Reading from UART timed out at byte 0!
21:31:49	[W]	[daikin_s21:257]	
Timeout waiting for RH response
21:31:49	[E]	[uart:015]	
Reading from UART timed out at byte 0!
21:31:49	[W]	[daikin_s21:257]	
Timeout waiting for RI response
21:31:49	[E]	[uart:015]	
Reading from UART timed out at byte 0!
21:31:49	[W]	[daikin_s21:257]	
Timeout waiting for Ra response
21:31:49	[E]	[uart:015]	
Reading from UART timed out at byte 0!
21:31:49	[W]	[daikin_s21:257]	
Timeout waiting for RL response
21:31:49	[E]	[uart:015]	
Reading from UART timed out at byte 0!
21:31:49	[W]	[daikin_s21:257]	
Timeout waiting for Rd response

[image]
[image]

wrouesnel, 2023-12-16:
What's your interface circuit look like? Do you have level shifters connected?

In my setup I have an M5Stack Lite, and I found I needed 10kOhm pull resistors between the RX/TX lines and the +5V.

hoangtridung, 2024-03-20:
> I'm using a standard [ESP32] board to connect Daikin's S21 (ftka20bv1h) with default config mentioned here: https://github.com/joshbenner/esphome-daikin-s21#configuration-example
> 
> but keep show the following error. Sending poweron commond is work but can't receive the state of the AC.
> .........
I have the same error. My circuit using ESP32 S2 mini with level shifters and  regulator to power ESP32 and shifters.

bjornenvall, 2024-05-30:
> What's your interface circuit look like? Do you have level shifters connected?
> 
> In my setup I have an M5Stack Lite, and I found I needed 10kOhm pull resistors between the RX/TX lines and the +5V.

Hi! can you explain, I'm struggling with a m5stack s3 lite. a resistor between what and where?

wrouesnel, 2024-05-30:
> > What's your interface circuit look like? Do you have level shifters connected?
> > In my setup I have an M5Stack Lite, and I found I needed 10kOhm pull resistors between the RX/TX lines and the +5V.
> 
> Hi! can you explain, I'm struggling with a m5stack s3 lite. a resistor between what and where?

I don't know what pin configuration you're using, but you'll have a +5V power supply line. That's the first point. The resistors you need should go from +5V to the RS-232 serial receive line, and the RS-232 serial trasmit line - 2 separate 10 kiloohm resistors.

Since transmit is working, you probably only need 10kOhm from +5V to the RS-232 serial receive line. You might be able to do it by stripping a small section of your control wires, though to do this I got some prototype board.

bjornenvall, 2024-05-31:
> I don't know what pin configuration you're using, but you'll have a +5V power supply line. That's the first point. The resistors you need should go from +5V to the RS-232 serial receive line, and [...]

Sorry but i still don’t understand. My ac only provide me with 14v so I have a stepdown thing (buck something) connected to atom 5v pin. The ground and the other two I have connected directly to the atom G and pin 1 and 2

wrouesnel, 2024-05-31:
> Sorry but i still don’t understand. My ac only provide me with 14v so I have a stepdown thing (buck something) connected to atom 5v pin. The ground and the other two I have connected directly to the [...]

Right, so you need to put a 10 kiloOhm resistor connected between the 5V pin of the atom, to pin 1 of the AC line - and another one connected from the 5V pin of the atom to pin 2 of the atom.

wrouesnel, 2024-06-02:
Yes

hoangtridung, 2024-06-03:
I used a board ESP32-C3 Supermini. My heatpump is FTXS serries, When set mode to Auto / HeatCool the heatpump are acknowledge but on the frontend of HA, it shows as OFF . Other mode setting is OK.
Any one have that error ???
My wiring as follow:

**- PIN 4, 5 on S21 connector → DC-DC buck → PIN Gnd, VCC on ESP32-C3**
**- PIN 2, 3 on S21 connector →  Shift Level → PIN 20,21 on ESP32-C3**

and coding are:
```
uart:
  - id: s21_uart
    tx_pin: GPIO20
    rx_pin: GPIO21
    baud_rate: 2400
    data_bits: 8
    parity: EVEN
    stop_bits: 2
daikin_s21:
  tx_uart: s21_uart
  rx_uart: s21_uart
```

reicharb, 2024-08-21:
> What's your interface circuit look like? Do you have level shifters connected?
> 
> In my setup I have an M5Stack Lite, and I found I needed 10kOhm pull resistors between the RX/TX lines and the +5V.

Are you using the M5 Stack Atom lite with the Atom rs-485 kit?  I have been trying to get it to work.  I am able to control the unit but I don't get any info back from the unit.

wrouesnel, 2024-08-21:
> Are you using the M5 Stack Atom lite with the Atom rs-485 kit? I have been trying to get it to work. I am able to control the unit but I don't get any info back from the unit.

No kits, it's just a bare M5 Atom Lite + the resistors.

chuc2rk, 2024-12-02:
> I used a board ESP32-C3 Supermini. My heatpump is FTXS serries, When set mode to Auto / HeatCool the heatpump are acknowledge but on the frontend of HA, it shows as OFF . Other mode setting is OK. [...]

does esp32-c3 work ok?

wrouesnel, 2024-12-02:
> does esp32-c3 work ok?

You might need a 10K pull-up resistor on the receive line between your level shifter and the +5V supply (see my posts above - but this was absolutely the problem I had that manifested exactly like this).

Your level shifter might have a provision to enable this already.

chuc2rk, 2024-12-03:
> Your level shifter might have a provision to enable this already.

Thanks! I will try for my DAIKIN FTKB35YVMV

[image]