joshbenner/esphome-daikin-s21 #20: What's the difference between using Faiking board and an ESP?

txitxo0, 2024-04-17:
Hi there,

I don't understand what is Faiking board doing. It is converting the Vcc to 5volts and wiring the S21 tx and Rx port to the esp?

Thank you

amzaldua, 2024-04-18:
I have it working with an ESP32-C3 Supermini and a downshifter HW 0108 using port 1 (5V) as power supply. My air conditioner is a Daikin FTXM35R5V1B.

txitxo0, 2024-04-18:
> Hola,
> I have it working with an ESP32-C3 Supermini and a logic converter HW0108 using port 1 (5V) as power supply. My air conditioner is a Daikin FTXM35R5V1B.

Thank you!

orthobots, 2024-04-19:
> Hola, I have it working with an ESP32-C3 Supermini and a logic converter HW0108 using port 1 (5V) as power supply. My air conditioner is a Daikin FTXM35R5V1B.

Hi. I'm planning to try this with a ESP32 WROOM. My S21 has got 5v VCC, so just wondering, why is the Logic Converter needed? 

Thanks

amzaldua, 2024-04-19:
> > Hola, I have it working with an ESP32-C3 Supermini and a logic converter HW0108 using port 1 (5V) as power supply. My air conditioner is a Daikin FTXM35R5V1B.
> 
> Hi. I'm planning to try this with a ESP32 WROOM. My S21 has got 5v VCC, so just wondering, why is the Logic Converter needed?
> 
> Thanks

Because the UART ports from Daikin also work with 5V. It is needed to reduce it to 3.3V.

MassiPi, 2024-04-19:
> Because the UART ports from Daikin also work with 5V. It is needed to reduce it to 3.3V.

it is needed only for certain models.
for many others (almost all of them) also 3.3V level is ok.
however, easily you may need a step down to power the ESP, sinche the +5V of Daikin split is commonly too weak

amzaldua, 2024-04-19:
The best thing is to measure the voltage provided by the communication pins with a multimeter to dispel any doubts. In my Perfera split unit, I got a reading of 5 volts.

orthobots, 2024-04-20:
> The best thing is to measure the voltage provided by the communication pins with a multimeter to dispel any doubts. In my Perfera HVAC unit, I got a readings of 5 volts.

Did yours work without the logic converter?

MassiPi, 2024-04-20:
you probably can in any case avoid logic converter and just use open drain output, if your split does not accept 3.3V logic level.

txitxo0, 2024-04-20:
> you probably can in any case avoid logic converter and just use open drain output, if your split does not accept 3.3V logic level.

But in the other way, maybe 5v from the split to the ESP is maybe too much power.

amzaldua, 2024-04-20:
> > Because the UART ports from Daikin also work with 5V. It is needed to reduce it to 3.3V.
> 
> @amzaldua Can you please describe wich PIN needs to be soldered where ?
> 
> My guess: S21 VCC -> HW0108 VCCB S21 VCC -> ESP32 5V S21 GND -> HW0108 GND S21 GND -> ESP32 GND S21 TX -> HW0108 B1 HW0108 A1 -> ESP32 RX S21 RX -> HW0108 B2 HW0108 A2 -> ESP32 TX HW0108 OE -> ESP32 3.3V HW0108 VCCA -> ESP 3.3V Is this correct ?

[image]

txitxo0, 2024-04-20:
> [image]

Why are you using a resistor in pulldown in B8? Thank you

amzaldua, 2024-04-20:
> > you probably can in any case avoid logic converter and just use open drain output, if your split does not accept 3.3V logic level.
> 
> But in the other way, maybe 5v from the split to the ESP is maybe too much power.

As MassiPi mentions, it also works without a logic converter, but in the long run, I'm not sure if it will end up burning the ESP32 since in my case, when measuring the voltage of the communication ports, I obtained up to 5V, and I prefer not to stress the ESP32. Perhaps MassiPi has taken measurements on their S21 port and observed lower voltages, but that's not my case.

amzaldua, 2024-04-20:
> > > > Because the UART ports from Daikin also work with 5V. It is needed to reduce it to 3.3V.
> > > 
> > > 
> > > @amzaldua Can you please describe wich PIN needs to be soldered where ?
> > > My guess: S21 VCC -> HW0108 VCCB S21 VCC -> ESP32 5V S21 GND -> HW0108 GND S21 GND -> ESP32 GND S21 TX -> HW0108 B1 HW0108 A1 -> ESP32 RX S21 RX -> HW0108 B2 HW0108 A2 -> ESP32 TX HW0108 OE -> ESP32 3.3V HW0108 VCCA -> ESP 3.3V Is this correct ?
> > 
> > 
> > ![IMG_4678](https://private-user-images.githubusercontent.com/116205677/324171480-a24dcc2f-b50b-458d-92fd-d240390843ce.jpeg?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MTM2MDk1ODYsIm5iZiI6MTcxMzYwOTI4NiwicGF0aCI6Ii8xMTYyMDU2NzcvMzI0MTcxNDgwLWEyNGRjYzJmLWI1MGItNDU4ZC05MmZkLWQyNDAzOTA4NDNjZS5qcGVnP1gtQW16LUFsZ29yaXRobT1BV1M0LUhNQUMtU0hBMjU2JlgtQW16LUNyZWRlbnRpYWw9QUtJQVZDT0RZTFNBNTNQUUs0WkElMkYyMDI0MDQyMCUyRnVzLWVhc3QtMSUyRnMzJTJGYXdzNF9yZXF1ZXN0JlgtQW16LURhdGU9MjAyNDA0MjBUMTAzNDQ2WiZYLUFtei1FeHBpcmVzPTMwMCZYLUFtei1TaWduYXR1cmU9OTAyMWEwYjExNTcxYTFhYWMwNWJkOWNjOTg3MjMzYWFmMTlhZmNhNTg4NGIwNjUxNzI3OGNhMWQzMTZmZTBmMSZYLUFtei1TaWduZWRIZWFkZXJzPWhvc3QmYWN0b3JfaWQ9MCZrZXlfaWQ9MCZyZXBvX2lkPTAifQ.hbxgGUj27I5g6ZizeMZ9e1JSkfMCg6TS7Mk1GXUj5Tc)
> 
> Why are you using a resistor in pulldown in B8? Thank you

The resistance is used in the EO port. Sorry for the photo perspective issue. This is a 1K resistor. It may not be necessary, but it is recommended to have it in some forums.

MassiPi, 2024-04-20:
what do you mean with "in the other way"? 
powering the esp has nothing to do with the logic level you are using
as i i wrote above, you easily need a stepdown to power the ESP (my splits don't even have a +5V power line available, only +14.5V)

amzaldua, 2024-04-20:
@txitxo0 ,  it seems we are confusing two things here. The pinout of the S21 port is as follows:
   • 1 - VCC (5V)
   • 2 - TX (5V)
   • 3 - RX (5V)
   • 4 - VCC (14,5V!!)
   • 5 - GND
The logic converter is for communication ports 2 and 3 in case they operate at 5V; if not, you don't need it.
For power supply, you can use ports 1 or 4 depending on the result of measuring both ports with a multimeter. If port 1 is disabled, you'll have to use port 4, but since it has more than 5V, you'll need to use a step-down converter to reduce the voltage to 5V. If, on the other hand, port 1 gives you 5V, as is my case, you can use it directly with your ESP32.

amzaldua, 2024-04-20:
@MassiPi , are you using this repository? If so, are you experiencing any issues? For example, I've encountered some communication errors. They don't seem significant because communication is functional, but sometimes it behaves strangely.
```
[16:06:37][D][sensor:094]: 'My Daikin Inside Temperature': Sending state 25.00000 °C with 1 decimals of accuracy
[16:06:37][D][sensor:094]: 'My Daikin Outside Temperature': Sending state 25.00000 °C with 1 decimals of accuracy
[16:06:37][D][sensor:094]: 'My Daikin Coil Temperature': Sending state 24.50000 °C with 1 decimals of accuracy
[16:06:37][D][sensor:094]: 'My Daikin Fan Speed': Sending state 0.00000 rpm with 0 decimals of accuracy
[16:06:38][D][daikin_s21.climate:312]: Room temp from external sensor: 24.3 °C (24.3 °C)
[16:06:38][D][daikin_s21.climate:313]:   Offset: 0.7
[16:06:38][D][climate:396]: 'My Daikin' - Sending state:
[16:06:38][D][climate:399]:   Mode: OFF
[16:06:38][D][climate:401]:   Action: OFF
[16:06:38][D][climate:407]:   Custom Fan Mode: Automatic
[16:06:38][D][climate:416]:   Swing Mode: HORIZONTAL
[16:06:38][D][climate:419]:   Current Temperature: 24.30°C
[16:06:38][D][climate:425]:   Target Temperature: 26.00°C
[16:06:39][W][component:237]: Component daikin_s21 took a long time for an operation (610 ms).
[16:06:39][W][component:238]: Components should block for at most 30 ms.
[16:06:41][W][component:237]: Component daikin_s21 took a long time for an operation (609 ms).
[16:06:41][W][component:238]: Components should block for at most 30 ms.
[16:06:43][W][component:237]: Component daikin_s21 took a long time for an operation (609 ms).
[16:06:43][W][component:238]: Components should block for at most 30 ms.
```

txitxo0, 2024-04-20:
> what do you mean with "in the other way"? 
> powering the esp has nothing to do with the logic level you are using
> as i i wrote above, you easily need a stepdown to power the ESP (my splits don't even have a +5V power line available, only +14.5V)

Yep, I know. I mean tx an RX in the S21 side is operating at 5 while the esp Rx/tx pins operates at 3.3v, so the logic converter to step down from 5 to 3.3 between S21 and esp sounds like sooo better way than to use directly 5v on those ports.

And apart of that, for power esp, we need 5v, if we have some Vcc port in S21 just linking to S21, if not a converter from 14v to 5v it is needed

txitxo0, 2024-04-20:
> @txitxo0 ,  it seems we are confusing two things here. The pinout of the S21 port is as follows:
>    • 1 - VCC (5V)
>    • 2 - TX (5V)
>    • 3 - RX (5V)
>    • 4 - VCC (14,5V!!)
>    • 5 - GND
> The logic converter is only for communication ports 2 and 3 in case they operate at 5V; if not, you don't need it.
> For power supply, you can use ports 1 or 4 depending on the result of measuring both ports with a multimeter. If port 1 is disabled, you'll have to use port 4, but since it has more than 5V, you'll need to use a step-down converter to reduce the voltage to 5V. If, on the other hand, port 1 gives you 5V, as is my case, you can use it directly with your ESP32.

Totally agree

MassiPi, 2024-04-20:
> tx an RX in the S21 side is operating at 5 while the esp Rx/tx pins operates at 3.3v

ESP devices' gpios are known to be 5V tolerant, so a level shifter is not strictly needed on ESP side.
it COULD be needed if you are trying to use one of the few Daikin models that are not 3.3V tolerant. In the long list of compatibility on the Faikin project page, i think there is just one.

> @MassiPi , are you using this repository?

no i'm not, sorry, i wrote my custom code (you can see it in my repositories..)

amzaldua, 2024-04-20:
> no i'm not, sorry, i wrote my custom code (you can see it in my repositories..)

I'm familiar with your work from the Home Assistant Forum. I've based my configuration on your advice. I use this ESPHome repository because of its straightforward installation process. How can I install your development on my ESP32?

amzaldua, 2024-04-20:
> ESP devices' gpios are known to be 5V tolerant, so a level shifter is not strictly needed on ESP side.
> it COULD be needed if you are trying to use one of the few Daikin models that are not 3.3V tolerant. In the long list of compatibility on the Faikin project page, i think there is just one.

The official website clearly states that the maximum voltage supported by all pins on ESP32 microcontrollers is 3.6V. Could it work with a higher voltage? It's possible, but there is no guarantee that the device won't eventually break.

MassiPi, 2024-04-20:
well it seems that's not the case :)
https://www.letscontrolit.com/forum/viewtopic.php?t=8845

amzaldua, 2024-04-20:
> well it seems that's not the case :) https://www.letscontrolit.com/forum/viewtopic.php?t=8845

Obviously, even though the manufacturer specifies 3.6V as the maximum limit, a safety margin is always included. However, these safety margins are set for a reason. For hacking a high-voltage device, I prefer to work within the manufacturer's specified parameters. For minor home experiments, it's good to know that I can reach up to 5V.

amzaldua, 2024-04-23:
@MassiPi I am testing a Mini560 step-down in another split Daikin, but it doesn't provide the correct output. It works with a 9V cell, but when connected to the 12V S21 pin, it doesn't provide power. Have you experienced this issue with your ESP8266 + Mini560 configuration?

amzaldua, 2024-04-23:
Pending confirmation, I believe the issue is that the GND was taken from pin 5 and the power from the mini560. 🤦‍♂️

MassiPi, 2024-04-23:
wait, not sure i'm getting what you say. You have to power the mini560 with Vcc and Gnd (2 pins from the S21 port). Then you power the ESP with Vcc and Gnd (2 pins from the mini560)
all GND must be shared otherwise nothing is gonna work.

closed 2024-05-11 by txitxo0

chuc2rk, 2024-12-02:
> Hola, I have it working with an ESP32-C3 Supermini and a logic converter HW0108. I am using port 1 (5V) for the power supply. My air conditioner is a Daikin FTXM35R5V1B.

Please share code, thanks!

chuc2rk, 2024-12-02:
> Hola, I have it working with an ESP32-C3 Supermini and a logic converter HW0108. I am using port 1 (5V) for the power supply. My air conditioner is a Daikin FTXM35R5V1B.

please share code, I have esp32c3 too. Thanks!
DAIKIN FTKB35YVMV
[image]