joshbenner/esphome-daikin-s21 #21: High Latency in daikin_s21 Component Operations

amzaldua, 2024-04-22:
Hello,

I've been experiencing an issue with the `daikin_s21` component within my ESPhome setup, which seems to be taking an unusually long time to complete operations. According to my logs, this component is blocking for more than 600 ms, while the general recommendation is that components should block for no more than 30 ms.

Here are the relevant log entries:
```
[23:19:43][W][component:237]: Component daikin_s21 took a long time for an operation (611 ms).
[23:19:43][W][component:238]: Components should block for at most 30 ms.
[23:19:45][W][component:237]: Component daikin_s21 took a long time for an operation (608 ms).
[23:19:45][W][component:238]: Components should block for at most 30 ms.

```
This behavior is consistent and replicable. It concerns me because such extensive blocking might affect the overall performance and responsiveness of the system. 

Could you please look into this? Any guidance on how to mitigate this issue or any planned updates that might address this would be greatly appreciated.

Thank you for your assistance and for the great work on the ESPhome components!

fusionstream, 2024-07-26:
My understanding is ESPHOME sent an update that makes these messages more visible than usual but they've always been present.

supertin, 2024-08-31:
See https://github.com/esphome/issues/issues/4717

Fix is to add
```
logger:
  logs:
    component: ERROR
```

I mean, it doesn't fix the slow component, but it will stop the logs being filled with warnings.

asund, 2025-03-16:
I have a pretty drastic fix for this https://github.com/joshbenner/esphome-daikin-s21/pull/33