joshbenner/esphome-daikin-s21 #24: esphome with esp32s6 UART presents errors

mx5gr, 2024-06-28:
As only ESP-IDF framework is supported by ESP32-C6, we defined the following esphome (2024.6.4) config:

`esphome:
  name: test-ac
  platformio_options:
    board_build.f_cpu: 160000000L
    board_build.f_flash: 80000000L
    board_build.flash_size: 4MB
    board_build.mcu: esp32c6
    build_flags: "-DBOARD_HAS_PSRAM"
    board_build.arduino.memory_type: qio_opi

esp32:
  board: esp32-c6-devkitc-1
  variant: ESP32C6
  framework:
    type: esp-idf
    version: 5.2.2    
    sdkconfig_options:
      CONFIG_ESPTOOLPY_FLASHSIZE_4MB: y    
    platform_version: 6.7.0    

external_components:
  - source: github://joshbenner/esphome-daikin-s21@main
    components: [ daikin_s21 ]

uart:
  - id: s21_uart
    tx_pin: GPIO16
    rx_pin: GPIO17
    baud_rate: 2400
    data_bits: 8
    parity: EVEN
    stop_bits: 2
    debug:

# The base UART communication hub.
daikin_s21:
  tx_uart: s21_uart
  rx_uart: s21_uart

climate:
  - name: Daikin Test
    platform: daikin_s21
    visual:
      temperature_step: 0.5`

When we tried to compile it, we get:
`In file included from src/esphome/components/daikin_s21/climate/daikin_s21_climate.cpp:3:
src/esphome/components/daikin_s21/climate/daikin_s21_climate.cpp: In member function 'virtual void esphome::daikin_s21::DaikinS21Climate::dump_config()':
src/esphome/components/daikin_s21/climate/daikin_s21_climate.cpp:27:22: error: format '%u' expects argument of type 'unsigned int', but argument 5 has type 'uint32_t' {aka 'long unsigned int'} [-Werror=format=]
   27 |   ESP_LOGCONFIG(TAG, "  Update interval: %u", this->get_update_interval());
      |                      ^~~~~~~~~~~~~~~~~~~~~~~
src/esphome/core/log.h:72:36: note: in definition of macro 'ESPHOME_LOG_FORMAT'
   72 | #define ESPHOME_LOG_FORMAT(format) format
      |                                    ^~~~~~
src/esphome/core/log.h:153:33: note: in expansion of macro 'esph_log_config'
  153 | #define ESP_LOGCONFIG(tag, ...) esph_log_config(tag, __VA_ARGS__)
      |                                 ^~~~~~~~~~~~~~~
src/esphome/components/daikin_s21/climate/daikin_s21_climate.cpp:27:3: note: in expansion of macro 'ESP_LOGCONFIG'
   27 |   ESP_LOGCONFIG(TAG, "  Update interval: %u", this->get_update_interval());
      |   ^~~~~~~~~~~~~
src/esphome/components/daikin_s21/climate/daikin_s21_climate.cpp:27:43: note: format string is defined here
   27 |   ESP_LOGCONFIG(TAG, "  Update interval: %u", this->get_update_interval());
      |                                          ~^
      |                                           |
      |                                           unsigned int
      |                                          %lu
src/esphome/components/daikin_s21/climate/daikin_s21_climate.cpp: In member function 'void esphome::daikin_s21::DaikinS21Climate::save_setpoint(float)':
src/esphome/components/daikin_s21/climate/daikin_s21_climate.cpp:132:12: error: enumeration value 'Disabled' not handled in switch [-Werror=switch]
  132 |     switch (mode) {
      |            ^
src/esphome/components/daikin_s21/climate/daikin_s21_climate.cpp:132:12: error: enumeration value 'Dry' not handled in switch [-Werror=switch]
src/esphome/components/daikin_s21/climate/daikin_s21_climate.cpp:132:12: error: enumeration value 'Fan' not handled in switch [-Werror=switch]
src/esphome/components/daikin_s21/climate/daikin_s21_climate.cpp: In member function 'esphome::optional<float> esphome::daikin_s21::DaikinS21Climate::load_setpoint(esphome::daikin_s21::DaikinClimateMode)':
src/esphome/components/daikin_s21/climate/daikin_s21_climate.cpp:156:10: error: enumeration value 'Disabled' not handled in switch [-Werror=switch]
  156 |   switch (this->s21->get_climate_mode()) {
      |          ^
src/esphome/components/daikin_s21/climate/daikin_s21_climate.cpp:156:10: error: enumeration value 'Dry' not handled in switch [-Werror=switch]
src/esphome/components/daikin_s21/climate/daikin_s21_climate.cpp:156:10: error: enumeration value 'Fan' not handled in switch [-Werror=switch]
src/esphome/components/daikin_s21/climate/daikin_s21_climate.cpp: In member function 'void esphome::daikin_s21::DaikinS21Climate::set_s21_climate()':
src/esphome/components/daikin_s21/climate/daikin_s21_climate.cpp:410:17: error: format '%s' expects argument of type 'char*', but argument 5 has type 'const esphome::LogString*' [-Werror=format=]
  410 |   ESP_LOGI(TAG, "  Mode: %s", climate::climate_mode_to_string(this->mode));
      |                 ^~~~~~~~~~~~
src/esphome/core/log.h:72:36: note: in definition of macro 'ESPHOME_LOG_FORMAT'
   72 | #define ESPHOME_LOG_FORMAT(format) format
      |                                    ^~~~~~
src/esphome/core/log.h:151:28: note: in expansion of macro 'esph_log_i'
  151 | #define ESP_LOGI(tag, ...) esph_log_i(tag, __VA_ARGS__)
      |                            ^~~~~~~~~~
src/esphome/components/daikin_s21/climate/daikin_s21_climate.cpp:410:3: note: in expansion of macro 'ESP_LOGI'
  410 |   ESP_LOGI(TAG, "  Mode: %s", climate::climate_mode_to_string(this->mode));
      |   ^~~~~~~~
src/esphome/components/daikin_s21/climate/daikin_s21_climate.cpp:410:27: note: format string is defined here
  410 |   ESP_LOGI(TAG, "  Mode: %s", climate::climate_mode_to_string(this->mode));
      |                          ~^
      |                           |
      |                           char*
Compiling .pioenvs/sens-1st-bedroom-ac/src/esphome/components/esp32/core.cpp.o
cc1plus: some warnings being treated as errors
*** [.pioenvs/sens-1st-bedroom-ac/src/esphome/components/daikin_s21/climate/daikin_s21_climate.cpp.o] Error 1
In file included from src/esphome/components/uart/uart.h:6,
                 from src/esphome/components/daikin_s21/s21.h:3,
                 from src/esphome/components/daikin_s21/s21.cpp:1:
src/esphome/components/daikin_s21/s21.cpp: In member function 'void esphome::daikin_s21::DaikinS21::check_uart_settings()':
src/esphome/components/daikin_s21/s21.cpp:107:11: error: format '%u' expects argument of type 'unsigned int', but argument 6 has type 'uint32_t' {aka 'long unsigned int'} [-Werror=format=]
  107 |           "  Invalid baud_rate: Integration requested baud_rate %u but you "
      |           ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
  108 |           "have %u!",
      |           ~~~~~~~~~~
src/esphome/core/log.h:72:36: note: in definition of macro 'ESPHOME_LOG_FORMAT'
   72 | #define ESPHOME_LOG_FORMAT(format) format
      |                                    ^~~~~~
src/esphome/core/log.h:149:28: note: in expansion of macro 'esph_log_e'
  149 | #define ESP_LOGE(tag, ...) esph_log_e(tag, __VA_ARGS__)
      |                            ^~~~~~~~~~
src/esphome/components/daikin_s21/s21.cpp:105:7: note: in expansion of macro 'ESP_LOGE'
  105 |       ESP_LOGE(
      |       ^~~~~~~~
src/esphome/components/daikin_s21/s21.cpp:108:18: note: format string is defined here
  108 |           "have %u!",
      |                 ~^
      |                  |
      |                  unsigned int
      |                 %lu
src/esphome/components/daikin_s21/s21.cpp: In member function 'virtual void esphome::daikin_s21::DaikinS21::dump_config()':
src/esphome/components/daikin_s21/s21.cpp:136:22: error: format '%u' expects argument of type 'unsigned int', but argument 5 has type 'uint32_t' {aka 'long unsigned int'} [-Werror=format=]
  136 |   ESP_LOGCONFIG(TAG, "  Update interval: %u", this->get_update_interval());
      |                      ^~~~~~~~~~~~~~~~~~~~~~~
src/esphome/core/log.h:72:36: note: in definition of macro 'ESPHOME_LOG_FORMAT'
   72 | #define ESPHOME_LOG_FORMAT(format) format
      |                                    ^~~~~~
src/esphome/core/log.h:153:33: note: in expansion of macro 'esph_log_config'
  153 | #define ESP_LOGCONFIG(tag, ...) esph_log_config(tag, __VA_ARGS__)
      |                                 ^~~~~~~~~~~~~~~
src/esphome/components/daikin_s21/s21.cpp:136:3: note: in expansion of macro 'ESP_LOGCONFIG'
  136 |   ESP_LOGCONFIG(TAG, "  Update interval: %u", this->get_update_interval());
      |   ^~~~~~~~~~~~~
src/esphome/components/daikin_s21/s21.cpp:136:43: note: format string is defined here
  136 |   ESP_LOGCONFIG(TAG, "  Update interval: %u", this->get_update_interval());
      |                                          ~^
      |                                           |
      |                                           unsigned int
      |                                          %lu
cc1plus: some warnings being treated as errors
*** [.pioenvs/sens-1st-bedroom-ac/src/esphome/components/daikin_s21/s21.cpp.o] Error 1
========================= [FAILED] Took 161.07 seconds =========================`

mx5gr, 2024-07-01:
I managed to compile the component by making the following alterations..

Within the climate/daikin_s21_climate.cpp file, find/replace:

void DaikinS21Climate::dump_config() {
  ESP_LOGCONFIG(TAG, "DaikinS21Climate:");
  ESP_LOGCONFIG(TAG, "  Update interval: %lx", this->get_update_interval());

(Note: %u becomes %lx)

Within the same file, three more conditions (Disabled/Fan/Dry) are added to the following:

void DaikinS21Climate::save_setpoint(float value) {
  auto mode = this->s21->get_climate_mode();
  optional<float> prev = this->load_setpoint(mode);
  // Only save if value is diff from what's already saved.
  if (abs(value - prev.value_or(0.0)) >= SETPOINT_STEP) {
    switch (mode) {
      case DaikinClimateMode::Auto:
        this->save_setpoint(value, this->auto_setpoint_pref);
        break;
      case DaikinClimateMode::Cool:
        this->save_setpoint(value, this->cool_setpoint_pref);
        break;
      case DaikinClimateMode::Heat:
        this->save_setpoint(value, this->heat_setpoint_pref);
        break;
      case DaikinClimateMode::Dry:
        //this->save_setpoint(value, this->heat_setpoint_pref);
        break;
	  case DaikinClimateMode::Fan:
        //this->save_setpoint(value, this->heat_setpoint_pref);
        break;
      case DaikinClimateMode::Disabled:
        //this->save_setpoint(value, this->heat_setpoint_pref);
        break;		
    }
  }
}

optional<float> DaikinS21Climate::load_setpoint(DaikinClimateMode mode) {
  optional<float> loaded;
  switch (this->s21->get_climate_mode()) {
    case DaikinClimateMode::Auto:
      loaded = this->load_setpoint(this->auto_setpoint_pref);
      break;
    case DaikinClimateMode::Cool:
      loaded = this->load_setpoint(this->cool_setpoint_pref);
      break;
    case DaikinClimateMode::Heat:
      loaded = this->load_setpoint(this->heat_setpoint_pref);
      break;
    case DaikinClimateMode::Dry:
      //loaded = this->load_setpoint(this->heat_setpoint_pref);
      break;
	case DaikinClimateMode::Fan:
      //loaded = this->load_setpoint(this->heat_setpoint_pref);
      break;
    case DaikinClimateMode::Disabled:
      //loaded = this->load_setpoint(this->heat_setpoint_pref);
      break;	  
  }
  return loaded;
}

Furthermore, within the s21.cpp file, find & replace the following:

void DaikinS21::check_uart_settings() {
  for (auto uart : {this->tx_uart, this->rx_uart}) {
    if (uart->get_baud_rate() != S21_BAUD_RATE) {
      ESP_LOGE(
          TAG,
          "  Invalid baud_rate: Integration requested baud_rate %u but you "
          "have %lx!",
          S21_BAUD_RATE, uart->get_baud_rate());

and

void DaikinS21::dump_config() {
  ESP_LOGCONFIG(TAG, "DaikinS21:");
  ESP_LOGCONFIG(TAG, "  Update interval: %lx", this->get_update_interval());
  this->check_uart_settings();
}

(Note: %u becomes %lx)

asund, 2024-12-28:
Do you actually want hex output? Some of those debug strings look like leaving them in decimal would be best. You might find the <cinttypes> header to be helpful for this problem. If I get some time over the holidays I can put up a PR that should play well with existing platforms.

closed 2025-03-01 by joshbenner