joshbenner/esphome-daikin-s21 #36: climate.CLIMATE_SCHEMA deprecated

benwa, 2025-06-02:
While building with ESPHome 2025.5, I see the following in the log:

```log
WARNING Using `climate.CLIMATE_SCHEMA` is deprecated and will be removed in ESPHome 2025.11.0. Please use `climate.climate_schema(...)` instead. If you are seeing this, report an issue to the external_component author and ask them to update it. https://developers.esphome.io/blog/2025/05/14/_schema-deprecations/. Component using this schema: daikin_s21
```

This looks to be used in [#L26](https://github.com/joshbenner/esphome-daikin-s21/blob/86ffe95fb5c0d6c7df440fd2672eb0d92d4d0955/components/daikin_s21/climate/__init__.py#L26)

txitxo0, 2025-07-17:
I have just noticed the same in the log. Is there any plan to update the component?

Thank you!

benwa, 2025-07-17:
Active development has moved to the https://github.com/asund/esphome-daikin-s21 fork